TauLabs/TauLabs #2013: Autotune on Naze32 produces incorrect values

dustin, 2015-10-24:
There've been a couple other reports of this, but I tried a couple of autotunes with various tweaks on naze32 last night and got unrealistically low taus.

mlyle, 2015-10-24:
need to determine if this affects CC3D also

tracernz, 2015-10-24:
Note: I have reports of this prior to #1990. ;) @dustin tested with the gyro rate at both 333 Hz and 500 Hz with the same result.

peabody124, 2015-10-24:
Could you test it on master (previous release) and see if it worked then. It would be nice to distinguish if it never worked v a regression. Also if anyone else can reproduce it to make sure it isn't a frame/board specific event.

VArcht, 2015-10-24:
Tested this on CC3D with taulabs_master_20150131, same result 0.00677 tau.

tracernz, 2015-10-24:
EMAX Nighthawk (My test rig)
Cobra 2204/1960 kv
Rotorgeeks 12A
DAL 5040
Dinogy 3S 1500 65C

Naze32
```
Measured properties
Gain	Bias	Tau(s)	Noise
Roll	9.55547	0.0169047	0.007	293.646
Pitch	9.67937	-0.00273166	0.007	205.503

Tuning aggressiveness
Damping	1.1
Noise sensitivity	1 %
Natural frequency	17.6

Computed Values
RateKp	RateKi	RateKd
Roll	0.00491452	0.0439584	1.93644e-05
Pitch	0.00434183	0.038836	1.71078e-05
Outer Kp	16.3843	-	-
Derivative cutoff	51
```

Sparky
```
Measured properties
Gain	Bias	Tau(s)	Noise
Roll	9.80113	0.0124977	0.0314	116.044
Pitch	9.39549	-0.016818	0.0314	92.09

Tuning aggressiveness
Damping	1.1
Noise sensitivity	1 %
Natural frequency	6.1

Computed Values
RateKp	RateKi	RateKd
Roll	0.00179425	0.00553777	4.14406e-05
Pitch	0.00269185	0.00830813	6.21718e-05
Outer Kp	5.67494	-	-
Derivative cutoff	20
```

The Sparky tune seems to fly well, the Naze tune obviously doesn't.

peabody124, 2015-10-24:
@tracernz Ok! That is hugely useful and confirms that naze/cc3d is definitely diverging and hitting the lower limit on tau

VArcht, 2015-10-24:
So, as requested, I reverted to taulabs_master_20141026 unfortunately about 30 secs into autotune the quad twitched to the right then after correcting flipped over to the right and piled into the asphalt, no useful data.

tracernz, 2015-10-25:
I measured the average dT_s as requested by @peabody124 using tracernz/TauLabs@a02d3b7621b7e4709d08fe9fa181c142a707a313. Additionally I tried setting the SystemIdent UAVO to manual update as currently it is sent every loop of the autotune task while it's tuning.

My results on the bench with UAVO = update on change (as per current next):
naze32 average dT_s = 4.88 ms,
sparky1 average dT_s = 2.85 ms
With the UAVO switched to manual update:
naze32 average dT_s = 4.28 ms

Finally I tried a real tune with the UAVO on manual update and the actual dT_s passed to af_predict instead of the constant DT_MS and it ended up being worse, especially on noise performance as you might expect.
```
Aircraft description: Naze32
EMAX Nighthawk (My test rig)
Cobra 2204/1960 kv
Rotorgeeks 12A
DAL 5040
Dinogy 3S 1500 65C

Observations: af_predict(X,P,u,y, dT_s);
Still obviously bad.
Average dt during tune was 4.334 s.

Measured properties
		Gain		Bias		Tau(s)	Noise
Roll	8.62834		0.016682	0.00676	2319.33
Pitch	7.61131		-0.00232935	0.00676	2411.97

Tuning aggressiveness
Damping				1.1
Noise sensitivity	1 %
Natural frequency	29.4

Computed Values
		RateKp		RateKi		RateKd
Roll	0.0287567	0.427737	0.000143318
Pitch	0.0795114	1.18268		0.000396271
Outer Kp 27.3553
Derivative cutoff 100
```

peabody124, 2015-10-26:
@tracernz if you have a chance could you try repeating this on https://github.com/TauLabs/TauLabs/releases/tag/20141026

tracernz, 2015-10-28:
CC3D @ 20141026 with the DT_s measuring code added: 3.17 ms average. That's a very definite difference. I tried building with the GCC version back then, and GCC 4.9.3, both gave similar results so the compiler is ruled out. The autotune module has been refactored quite a bit since then and I'm sure many many other things have changed so it will take a bit of work to figure out what's happened. 

Also of note is how much faster GCS was back then!! I didn't even get a chance to see the splash screen when loading whereas now it takes at least 10 seconds. Everything was a lot snappier in general.

VArcht, 2015-10-28:
Just a note, Endo and 20141026  GCS's launch in identical times here on windows. I launched them concurrently with a bat file and the complete within milliseconds of each other.

mlyle, 2015-11-02:
Is the difference between released and debug configuration builds, then?

dustin, 2015-11-08:
I'm getting bogus values on my sparky now.

```
Measured properties
Gain	Bias	Tau(s)	Noise
Roll	9.58036	-0.00280699	0.0241	536.017
Pitch	9.38116	-0.0264697	0.0241	163.054

Tuning aggressiveness
Damping	1.1
Noise sensitivity	1 %
Natural frequency	7.2

Computed Values
RateKp	RateKi	RateKd
Roll	0.00245846	0.00897029	4.28864e-05
Pitch	0.00300035	0.0109475	5.23394e-05
Outer Kp	6.70478	-	-
Derivative cutoff	23	-	-
```

http://forum.taulabs.org/viewtopic.php?f=27&t=268&start=120#p8176

peabody124, 2015-11-13:
Report this worked on CC3D #1680

peabody124, 2015-11-14:
Ok. Things are not so straightforward :(. Just got a CC3D flying again on a smaller quad.

[image]

That being said, it's way too twitchy on default settings with this.

peabody124, 2015-11-14:
On https://github.com/TauLabs/TauLabs/releases/tag/20141026 I get very similar values

[image]

dustin, 2015-11-14:
I tried a few autotunes on sparky tonight.

First, on endo using approximately my latest config for this thing:  Got 7.88ms

Then, I reset inner and outer loops and did it again and got around 11ms.

I went to next (plus my changes) and tried my base config.  Got ~7ms.

I tried resetting just the outer loop, got ~7ms.

Tried resetting the outer and inner loops and got ~7ms.

peabody124, 2015-11-15:
Yeah I can confirm on Sparky2 and next I get the same autotune as I do with endo and 20141025 with the same frame. On CC3D these also produce identical (but different values) as well as with next.

In addition, when running AT on CC3D (for really any of these) the CPU load hit 100%. With diagnostics enabled 33% of the CPU usage is going to AT alone. My leading suspicion right now is there is just too much FP math for an F1 too handle. I'm not sure increasing the step time will be ok either since really 3ms is pretty borderline for estimating things with a tau of 10ms.

peabody124, 2015-11-15:
More suspicious. I implemented a 2 axis version of the AT SystemIdent. This reduced the CPU usage on CC3D.

[image]

However still didn't produce a good fit. Then I tried reducing gyro rate (which brings down entire CPU load). You see that AT usage goes up

[image]

Which is consistent with there just not being enough CPU usage to keep up with the data stream. I'm pretty convinced the F1 processor just cannot handle this form of SI. There might be a way to make a more specific stream lined single axis version of this, maybe not even using an EKF but a fixed gain filter.

peabody124, 2015-11-15:
Dropping rate to 200 (previous above was 333) makes AT go up further:

[image]