TauLabs/TauLabs #2026: Negative climb rate during rapid ascent from 0

dustin, 2015-10-29:
I've noticed that my climb rate is calculated to be negative when taking off even though the altitude indicator is only increasing:

http://gfycat.com/GlassImpartialHydra

It seems to catch up and go positive again, but it's quite surprising to ever go negative in this scenario.

kubark42, 2015-10-29:
* Is the VSI the bottom right corner?
* What flight controller board is this? 
* What UAVO is being displayed there?

Agreed that it shouldn't go negative like that.

dustin, 2015-10-29:
This is BrainFPV.  Perhaps @BrainFPV would know more about the details.  I assumed this isn't board specific.

Yeah, the climb rate is in the bottom right.  It starts at zero, then goes negative as I start climbing rapidly, then crosses zero and goes positive.

BrainFPV, 2015-10-30:
Strange.. this shouldn't be Brain specific, as the [osd code](https://github.com/BrainFPV/TauLabs/blob/brain/flight/Modules/OnScreenDisplay/onscreendisplay.c#L1229) uses `VelocityActual.Down` directly.

mluessi, 2015-10-30:
Just discussed this with @kubark42. The effect you are seeing could be due to pressure changes caused by the props. Can you do a test where you hold down the quad an spin up the props to see if the altitude changes? Are you using foam on the barometer?

dustin, 2015-10-30:
There's foam on the baro. The altimeter is increasing the whole time.  It's
just the climb rate that starts out negative.

kubark42, 2015-10-30:
@dustin The barometer is filtered at different rates as it relates to the position and velocity estimates. If you can do a quick sanity test as @mluessi describes this well help us narrow down the possibilities. Based on the scale of the error in your video, it's an interesting problem.