TauLabs/TauLabs #2032: MSP_RC values are confusing with s.bus

dustin, 2015-10-31:
The OSD has stick commands for the following:

* Getting out of the stats screen after disarming
* Changing OSD HUDs
* Configuration (not yet implemented, but important)

With s.bus, the stick commands are sending 172 - 1811 instead of the expected 1000-2000 that will make the OSD happy.  These should be normalized.

tracernz, 2015-10-31:
There are normalised values for the main channels in the [ManualControlCommand UAVO](https://github.com/TauLabs/TauLabs/blob/next/shared/uavobjectdefinition/manualcontrolcommand.xml), perhaps scale them with `cmd * 5 + 1500` ?

mluessi, 2015-10-31:
I would use the min. and max. from `ManualControlSettings` to rescale all channels to the 1000.2000 range that the OSD expects.

dustin, 2015-11-01:
@mluessi thanks, I gave that a shot.  Still need real testing on a quad, but I think it looks right.

mlyle, 2015-11-01:
I prefer using manualcontrolcommand.pitch *500 + 1500 etc here... As that is already normalized properly and just needs moving to the 1000-2000 range (and handles input reversal correctly)

dustin, 2015-11-02:
@mlyle The downside of using the normalized values is that they're not available in an array and it's not available for all channels (AFIACT), so I've got to do more stuff.

This should do the right thing for all channels, including reversal and without using floating point math.  (I'm testing it in a few minutes).

mlyle, 2015-11-02:
Ok.  I am not horribly opposed, but I'm not also generally a fan of more things duplicating the rc interpretation logic in different subsystems in potentially slightly different ways.  I agree it's unfortunate that you can't use sub scripted access.

I think manualcontrolcommand and accessory desired have all the desired values, though.

dustin, 2015-11-02:
I've not tested it, but it should look something like this:

```
index 5b0e53e..e1b0b9d 100644
--- a/flight/Modules/UAVOMSPBridge/UAVOMSPBridge.c
+++ b/flight/Modules/UAVOMSPBridge/UAVOMSPBridge.c
@@ -35,7 +35,7 @@
 #include "flightbatterystate.h"
 #include "gpsposition.h"
 #include "manualcontrolcommand.h"
-#include "manualcontrolsettings.h"
+#include "accessorydesired.h"
 #include "attitudeactual.h"
 #include "airspeedactual.h"
 #include "actuatorsettings.h"
@@ -126,7 +126,6 @@ typedef enum {
 } msp_state;
 
 struct msp_bridge {
-	ManualControlSettingsData cmdSettings;
 	uintptr_t com;
 
 	msp_state _state;
@@ -148,7 +147,6 @@ struct msp_bridge {
 #define TASK_PRIORITY               PIOS_THREAD_PRIO_LOW
 
 static bool module_enabled;
-static volatile bool cmd_settings_updated = true;
 extern uintptr_t pios_com_msp_id;
 static struct msp_bridge *msp;
 static int32_t uavoMSPBridgeInitialize(void);
@@ -337,53 +335,33 @@ static void _msp_send_altitude(struct msp_bridge *m)
 	msp_send(m, MSP_ALTITUDE, data.buf, sizeof(data));
 }
 
-static uint16_t _msp_scale_rc(struct msp_bridge *m,
-			      ManualControlCommandData manualState,
-			      int channel_num)
-{
-	int val = manualState.Channel[channel_num];
-	int min = m->cmdSettings.ChannelMin[channel_num];
-	int max = m->cmdSettings.ChannelMax[channel_num];
-	int idle = m->cmdSettings.ChannelNeutral[channel_num];
-
-	// Min and max may be reversed.
-	if (min > max) {
-		int tmp = min;
-		min = max;
-		max = tmp;
-	}
-	// If the value is outside of the given range, return the
-	// neutral value.
-	if (val < min || val > max) {
-		return idle;
-	}
-
-	return (((val - min) * 1000) / (max - min)) + 1000;
+static uint16_t _msp_scale_rc(float percent) {
+	return percent*500 + 1500;
 }
 
 // MSP RC order is Roll/Pitch/Yaw/Throttle/AUX1/AUX2/AUX3/AUX4
 static void _msp_send_channels(struct msp_bridge *m)
 {
+	AccessoryDesiredData acc0, acc1, acc2;
 	ManualControlCommandData manualState;
 	ManualControlCommandGet(&manualState);
-	if (cmd_settings_updated) {
-		cmd_settings_updated = false;
-		ManualControlSettingsGet(&m->cmdSettings);
-	}
+	AccessoryDesiredInstGet(0, &acc0);
+	AccessoryDesiredInstGet(1, &acc1);
+	AccessoryDesiredInstGet(2, &acc2);
 
 	union {
 		uint8_t buf[0];
 		uint16_t channels[8];
 	} data = {
 		.channels = {
-			_msp_scale_rc(m, manualState, MANUALCONTROLCOMMAND_CHANNEL_ROLL),
-			_msp_scale_rc(m, manualState, MANUALCONTROLCOMMAND_CHANNEL_PITCH),
-			_msp_scale_rc(m, manualState, MANUALCONTROLCOMMAND_CHANNEL_YAW),
-			_msp_scale_rc(m, manualState, MANUALCONTROLCOMMAND_CHANNEL_THROTTLE),
-			_msp_scale_rc(m, manualState, MANUALCONTROLCOMMAND_CHANNEL_FLIGHTMODE),
-			_msp_scale_rc(m, manualState, MANUALCONTROLCOMMAND_CHANNEL_ACCESSORY0),
-			_msp_scale_rc(m, manualState, MANUALCONTROLCOMMAND_CHANNEL_ACCESSORY1),
-			_msp_scale_rc(m, manualState, MANUALCONTROLCOMMAND_CHANNEL_ACCESSORY2),
+			_msp_scale_rc(manualState.Roll),
+			_msp_scale_rc(manualState.Pitch),
+			_msp_scale_rc(manualState.Yaw),
+			_msp_scale_rc(manualState.Throttle),
+			_msp_scale_rc(acc0.AccessoryVal),
+			_msp_scale_rc(acc1.AccessoryVal),
+			_msp_scale_rc(acc2.AccessoryVal),
+			1000, // no aux4
 		}};
 
 	msp_send(m, MSP_RC, data.buf, sizeof(data));
@@ -556,11 +534,6 @@ static void setMSPSpeed(struct msp_bridge *m)
 	}
 }
 
-static void _msp_cmd_settings_update_cb(UAVObjEvent * ev)
-{
-	cmd_settings_updated = true;
-}
-
 /**
  * Module initialization routine
  * @return 0 when initialization was successful
@@ -580,7 +553,6 @@ static int32_t uavoMSPBridgeInitialize(void)
 			msp->com = pios_com_msp_id;
 
 			setMSPSpeed(msp);
-			ManualControlSettingsConnectCallback(_msp_cmd_settings_update_cb);
 
 			module_enabled = true;
 			return 0;
```

mluessi, 2015-11-02:
@dustin yes, I think this is a better solution. Let me manual control module do the work instead of replicating it here. For some reason I didn't think of the `AccessoryDesired` objects.

commit c98238b8 (2015-11-03): Scale sticks for MSP_RC response.

MSP wants 1000-2000 in practice.  s.bus values are quite a bit different
and caused a lot of confusion.

Fixes: #2032
closed 2015-11-03 by mlyle