TauLabs/TauLabs #2057: In-flight reset

dustin, 2015-11-14:
I've had a sparky 1 reset in flight due to the independent watchdog firing.  It seems to be easy to reproduce for me.

Example of the reboot happening in flight here:  http://gfycat.com/SelfassuredMenacingGourami

What you see there is flying, battery alarm goes off (not sure why) then independent watchdog caused a reboot.  This happened three times in ~half hour there.  Possibly happened before in a catastrophic crash that led me to writing code to communicate alarms to the OSD.

This particular flight was running something like this code:  https://github.com/dustin/TauLabs/tree/mspalrm  (tl/next + the work I've done for MSP alarm communication).  The MWOSD side of that is still sitting in a [PR queue](https://github.com/ShikOfTheRa/scarab-osd/pulls).  Basically merge my three changes.

I've got battery monitoring and MSP enabled.

Attached my config:

[f22pretest.txt](https://github.com/TauLabs/TauLabs/files/34487/f22pretest.txt)

dustin, 2015-11-14:
I increased the battery module stack by `128` (for ease of hackery, `#define STACK_SIZE_BYTES (496 + 128)`).  I managed to poke around and get the stack remaining value to `124`.  This seems like pretty red-flaggish.

mlyle, 2015-11-14:
Fixed by #2058.

closed 2015-11-14 by mlyle

dustin, 2015-11-15:
I just flew four batteries through my test aircraft that rebooted three times in two batteries last weekend.  I was not very gentle, but saw no reboot issues.  Had alarms firing all over the place, a few crashes, etc...  If I ever could repro, I can't now.

tracernz, 2015-11-15:
Nice work @dustin!