TauLabs/TauLabs #2059: Arm Settings are Decpetive - and dangerous!

SystemsGuy, 2015-11-14:
When setting "arming", the text on the page specifically states that you are selecting an option "+ low throttle".  This is not the case with switch arming, where you have to explicitly select "switch + low throttle".

I believe this creates a dangerous situation  Rather than have "Switch + Throttle" the selection in fact should be "Switch + IGNORE throttle" if you want to arm at a non-zero throttle value.  In reality this should probably be "not neutral" throttle as opposed to "low/high" so that at some point in the future we can support reverse throttle.
[image]

mluessi, 2015-11-18:
Why are we ignoring throttle when using switch arming in the first place? It seem very dangerous to me, i.e., you could have throttle on max, flip a switch and all motors spin up instantly..

I think we should add a throttle check to switch arming. This would also make it easy for reversible ESCs, where we could check that throttle is close to center stick.

tracernz, 2015-11-18:
@mluessi We already have that option, it's called "Switch+Throttle" or "Switch+Throttle+Delay". Is your suggestion to remove the "Switch" option which doesn't have a check?

DTFUHF, 2015-11-18:
I use the switch + no throttle safety, and much prefer it to switch+throttle especially with planes. Maybe it's less safe, but safety is the user's responsibility in all cases anyway.

SystemsGuy, 2015-11-19:
I'm not suggesting that it be removed - simply suggesting that the user should have to explicitly choose to ignore the throttle as opposed to having to explicitly choose to "not ignore" the throttle.  It would make the wording and description simpler as well.

tracernz, 2015-11-19:
AFAIK the default is "Always Disarmed" so the user always explicitly chooses the arming mode?

SystemsGuy, 2015-11-19:
True, but all of the options with the exception of switch arming respect the throttle setting - maybe it's just me, but it seems counterintutive for switch arming to be different.

My issue is that seeing the word "switch" to me implied "switch + low throttle" - which is not the case - if I'd looked further down and noticed the "switch + throttle" maybe I would have realized what was going on.  My suggestion is simply that we make it more obvious that selecting "switch" arming will allows you to arm at full throttle.

mlyle, 2015-11-19:
I think the main issue is that the label to the left of the pulldown is a lie for the switch setting.

tracernz, 2015-11-21:
The best solution I can think of is to edit that label and remove "throttle off", and add "+Throttle" to all the arming modes which use throttle, and maybe add a conditional label with a red warning when an arming mode that doesn't contain "Throttle" or "Always" is selected.