TauLabs/TauLabs #2070: Failed to Autotune Sparky 1

dustin, 2015-11-20:
I don't know if this is related to #2013, but I run at least four autotunes tonight.  All with default gains and no other modules enabled.  One on the latest release, and the rest on next.  Every one got < 7ms tau.

Example:

```
Autotune Results
 by dlsspy » Thu Nov 19, 2015 7:34 pm

Flight controller: Sparky
Firmware tag: mspalrm
Firmware commit: bcdeb7f
Firmware date: 20151120 03:06

Aircraft description: MRSS 180 - 22 with littlebee 20A ESCs, CM2204/2300kv motors. Just trying to reset to defaults and fly with 4x4x3 props. Did not autotune correctly.	

Observations: Seemed to do the right thing. Kept wanting to yaw right. Got bad answer.

Measured properties
Gain	Bias	Tau(s)	Noise
Roll	10.0514	-0.000791705	0.00674	197.923
Pitch	10.0987	-0.0179036	0.00674	83.7532

Tuning aggressiveness
Damping	1.1
Noise sensitivity	1 %
Natural frequency	21.1

Computed Values
RateKp	RateKi	RateKd
Roll	0.0039134	0.0418796	1.7259e-05
Pitch	0.0037327	0.0399458	1.6462e-05
Outer Kp	19.633	-	-
Derivative cutoff	65	-	-
```

fujin, 2015-11-22:
```
01:36 <peabody124> dsal: your quad you keep getting 7ms on, i think you said it was pretty well tuned for speed. let’s humor it and see how fast it will fix
01:37 <peabody124> can you try changing autotune task to run at 1ms
01:37 <peabody124> and then change this check: https://github.com/TauLabs/TauLabs/blob/next/flight/Modules/Autotune/autotune.c#L512 to cap at -6 instead of -5
01:37 <peabody124> see what time constant you get then. if it keeps stably hitting ~7ms maybe that is true ?
```

mluessi, 2015-11-23:
IMO, I would be somewhat surprising if the 7ms value is real. It would mean that for some reason the latency using F3/MPU9150 is much lower than on other targets. How well does it handle using the computed PID parameters?

mlyle, 2015-11-23:
I think the idea isn't that it's 7ms, but the intervals and "clipping" against the 7ms bound prevent us from converging to a valid sub-12ms solution.

dustin, 2015-11-24:
I didn't try applying it.  My previous tune was something like 18ms.  The only thing I changed for this one was the props (from 4045BN to 4x4x3).  I didn't actually try applying as it seems quite obviously bogus.  Haven't tried new settings yet, either.