TauLabs/TauLabs #2072: User injured by incorrect behavior of arming switch

mlyle, 2015-11-21:
```
14:14 < droidicus> hey all, I have a report of a potentially dangerous corner 
                   case that a friend just experienced...
14:14 < droidicus> Was working on the bench and had the quad in "never arm" mode
14:16 < droidicus> we back to change that setting, and accidently chose "arming 
                   switch" instead of holding the yaw at max (cannot remember 
                   what that is called in the software)
14:17 < droidicus> however, he didn't have a arming switch set in his normal 
                   config
14:17 < droidicus> the quad immediately armed and flew at him
14:17 < droidicus> now he is in the ER getting the end of his finger sewn back on 
                   :-/
14:17 < dsal> That sounds like a bug to file in short order.
```

If an arming switch is not configured, and arming mode switch is selected, the behavior is the same as "always armed" mode.

SystemsGuy, 2015-11-21:
Confirmed "Amputate Mode" on Endo release. Configure PPM input, 5 channels with no arm, and select "Switch" as your arming method, and you are armed on power up. The FC is passing throttle to the ESC as it boots if you power on with throttle on your Rx. Only your ESC "low throttle arm" is protecting you here.

mlyle, 2015-11-21:
Insufficient check was introduced in 10b470b5.

SystemsGuy, 2015-11-22:
Tau Labs (detached from 106ce7d): 20151121-106ce7da appears to correct this issue, but intriduces odd behavior.

GCS is unable to save config after Radio Wizard - eg "red X" on Save button without arm settings set.
GCS is says unable to save config after setting arm settings, but appears to actually be saving them.

PR #2073 (2015-11-22): TransmitterControl: do not allow switch arming when invalid
fixes #2072 

needs test and validation.
closed 2015-11-22 by peabody124