TauLabs/TauLabs #2074: Automatically determine timer resolution for PWM outputs

mluessi, 2015-11-22:
We currently have a selection for timer resolution (1MHz/12MHz) in the outputs section. I think this is confusing to users, as most of them don't understand what the timer resolution is / does and why it is important. Furthermore, the 1MHz/12MHz choice is closely related to the clock frequencies of the currently used STM32 chips and may constrain us in the future.

I propose that we remove the timer resolution combo boxes and let the firmware determine the optimal resolution for a given PWM frequency.

peabody124, 2015-11-22:
Also middle ground: the firmware representation could stay the same, but the GCS could transparently make that decision and abstract it away. In theory there are some times you might want to manually determine the resolution (e.g. 1mhz has lower load at 400Hz, but in principle 12mhz can work too).

kubark42, 2015-11-22:
Agreed that GCS is a better place to do it, so long as there's no strong argument for in general reducing the number of configuration options.

mlyle, 2015-11-22:
It doesn't address one of @mluessi's concerns-- the hardware may know better what's actually attainable.  It's unfortunate to have to plumb a bunch of logic to handle something else with an 8MHz timer on the GCS side.

kubark42, 2015-11-22:
Agreed, but having used these boards for strange and unusual things I appreciate their flexibility. Would be a shame to lose that if not necessary. Could also be that the hardware informs the GCS of what it can do (via the board manager, for instance), and the GCS selects the best choice. In a distantly similar manner, this is what we do for timer banks.

tracernz, 2015-11-22:
I like this idea! It's a common query from basically everybody that comes along. Perhaps we can keep the existing options in the UAVO and add a new "auto" option that is the default, and remove the dropdown in the motors tab. That would allow the firmware to auto-select the frequency unless the user specifically overrides it in the UAVO browser.

mluessi, 2015-11-22:
I don't see how it would be better for the GCS to decide this. After all it's not very difficult to determine and if the GCS has to make the decision, the target needs to somehow inform the GCS what timer frequencies / resolutions it can support, which adds just unnecessary complexity.

mluessi, 2015-11-22:
I like the "auto option" suggest by @tracernz, but somehow the target still would need to inform the GCS what timer resolutions it can support.

mlyle, 2015-11-22:
I agree with @mluessi.  Also, if we select "wrongly" we're only likely to be off in the direction of wasting a little CPU, which is surely OK?

kubark42, 2015-11-22:
I can get behind the "auto" solution. What's important to me is to be able to have the granularity to set it if the need arises. This could be as easy as a uint8 which when set to 0 is "auto" and then otherwise the firmware tries to match the MHz.

P.S. FWIW, the target is already informing the GCS about the timer banks, so it would be very easy to extend this to support frequencies as well. Still, I like @tracernz's suggestion as an even better middle middle-ground.

mluessi, 2015-11-22:
@kubark42 it's not actually the firmware informing the GCS about the timer banks, it's the board plugin. But I guess it could be done the same way.

That being said, what is the actual use case where you would want to set the timer to be a lower resolution than the hardware supports? This is the only case that would be covered by the manually setting the resolution.

kubark42, 2015-11-23:
Completely fair. I meant that the files added to TL as part of adding a target include this info. 

Can't think of one right now, so it's probably not that great a problem/feature. 

One thing to consider is that there should be a UAVO which reports the auto-determined speed. This info should make its way back up to the telemetry stack so that we can always be sure the autodetection routine is working as intended.

doudar, 2016-02-20:
@mluessi Don't ask why, but I'm currently running different ESC's on 1-2 than I am on 3-4. My 1-2 ESC's support 12mhz & OneShot, 3-4 do not. Anything other than identical settings (I have to use the lowest common denominator, 400 & 1mhz) for all 4 channels causes random motor cutout and throttle up during flight.

mlyle, 2016-02-20:
@doudar-- this isn't to autonegotiate with the ESC.  It's to let the flight controller figure out the timer resolution based on the pulserate, maximum, and minimum pulse lengths.  If you configure those the same, you'll get identical PWM on the other channels.