TauLabs/TauLabs #2080: Ensure that defaultname in UAVO is present in list of options

kubark42, 2015-11-23:
E.g., ```<field name="FirstMotorSense" units="" type="enum" elements="1" options="CW,CCW" defaultvalue="None"/>``` passes the generator just fine. The generator should catch this and fail.

tracernz, 2015-11-24:
This is a little difficult to do with the new inheritance system introduced with HAL as the field parents are not resolved in the parser. We don't know what the options are for inherited fields who don't specify their own options, e.g. https://github.com/TauLabs/TauLabs/blob/next/shared/uavobjectdefinition/hwaq32.xml#L108.

mlyle, 2015-11-24:
Could easily be done at resolveParents time

tracernz, 2015-11-24:
True @icee. I managed to have a battle with whitespace :<, ended up cleaning up the whole file inadvertently.

PR #2083 (2015-11-24): UAVOGen: Check that default values for enums actually exist
Checks in options list to ensure the default values are actually possible.
Fixes #2080

Apologies for the extra diff due to whitespace cleanup. I made a mess of my own diff (tabs instead of spaces) and told the editor to fix it, and it turns out the rest of the file had some problems already. I think fixing it is the right move anyway but can revert those lines if necessary.
commit cd762773 (2015-11-24): UAVOGen: Check that default values for enums actually exist
Checks in options list to ensure the default values are actually possible.
Fixes #2080
closed 2015-11-24 by mlyle