TauLabs/TauLabs #2099: ChibiOS NIL port?

beriberikix, 2015-11-28:
Has anyone attempted to port TL to use [NIL](http://chibios.org/dokuwiki/doku.php?id=chibios:product:nil:start), from ChibiOS? It is smaller and uses less resources, though has [less features](http://chibios.org/dokuwiki/doku.php?id=chibios:articles:rt_vs_nil) than the full RTOS. As long as TL doesn't use some of the more advanced features of the RTOS, it should be do-able.

My end-goal is to get TL running on [Cheerson CX-10](http://www.eevblog.com/forum/reviews/teardown-cheerson-cx10-mini-quadcopter/).

tracernz, 2015-11-28:
It doesn't have mutexes, queues or dynamic memory, not going to be easy since those are all used extensively throughout the firmware.

beriberikix, 2015-11-28:
Shucks, figure that might have been the case :/

mluessi, 2015-12-02:
Closing this, as NIL doesn't have the features we need.

closed 2015-12-02 by mluessi