TauLabs/TauLabs #2105: ubx_cfg missing 'usage' byte in CFG_SBAS message

droney, 2015-12-09:
I was reviewing the u-blox M6 and M8 specs and it appears the 'usage' byte is missing from the UBLOX_CFG_SBAS msg construction in ubx_cfg_set_sbas(). The 'usage' byte should appear between the mode and number of SBAS tracking channels parameters. This omission obviously skews the remaining message contents.

Please refer to the u-bloxM8_ReceiverDescrProtSpec_(UBX-13003221)_Public.pdf document for specifics on the CFG-SBAS configuration.

mlyle, 2015-12-09:
I think the comments don't line up but it is correct.
```
    const uint8_t msg[] = {
        UBLOX_CFG_CLASS, // CFG
        UBLOX_CFG_SBAS,  // SBAS
        0x08,            // length lsb
        0x00,            // length msb
0        enable,          // enable flag
1       0b011,           // mode
2        3,               // # SBAS tracking channels
3        0,
4        sv_mask,
5        sv_mask >> 8,
6        sv_mask >> 16,
7        sv_mask >> 24,
```

I think the "mode" here corresponds to the "usage" field in current docs (I copied this from docs before so I wonder if the field name has changed), enabling ranging and differential corrections.  The rest of the fields line up, e.g. bytes 3-7 for satellite mask, bytes 2 for the number of tracking channels.

droney, 2016-01-14:
only comments are off.... closing issue

closed 2016-01-14 by droney