TauLabs/TauLabs #2110: Firmware linking against wrong UAVO library

kubark42, 2015-12-15:
Could be a consequence of https://github.com/TauLabs/TauLabs/pull/1985, but I have seen a problem several times now where the firmware links against the wrong library. Not sure how repeatable this is, but it last occurred to me after building our in-house board firmware on it current branch, switching to #2108 to effect changes, and switching back to our in-house board support branch.

My current workaround is simply to delete the uavo-synthetics directory and rebuild. It can be somewhat hard to link the build error to this fix because it only occurs when an already built file is rebuilt, so it looks like things magically broke (which was my first indication the make process might be responsible).

@mlyle, have you seen this?

mlyle, 2015-12-15:
https://github.com/d-ronin/dRonin/pull/24

mlyle, 2015-12-15:
Note that unrelated to recent changes to UAVOs there are many categories of dependency problems on the GCS side, some of which we've fixed in dRonin.

kubark42, 2015-12-18:
@mlyle d-ronin/dRonin#24 did not resolve the problem.

mlyle, 2015-12-18:
Do you have a repro scenario and platform?

We've been successfully toggling between Tau & dRonin successfully since that change.

kubark42, 2015-12-18:
I'll see if I can find one that uses public code. It's between private branches. It might be that the dRonin fix is only valid when switching between branches which both have the fix, although from your comment I suspect that might not be the case.

mlyle, 2015-12-18:
No, both sides need the fix.  Otherwise the one without the fix won't do the right thign.

kubark42, 2015-12-18:
That explains the problem, then. It's going back and forth to a branch from early September that causes the problem.

kubark42, 2015-12-23:
@mlyle I had a build "success" where the wrong UAVO was linked in and the firmware still compiled and ran. The result was the flight vehicle doing flips until the old library was manually deleted. This was using https://github.com/d-ronin/dRonin/pull/24. While this in theory solves the problem between two branches which both have https://github.com/d-ronin/dRonin/pull/24, can you think of a way to do it so that the newer version can never accidentally use the old version?

mlyle, 2015-12-23:
I recommend just doing clean builds.  Flight is mostly free of dependency issues now, but ground/gcs are not-- both in various kinds of static dependencies between plugins and missed dependencies between build steps.

make all_clean && make -j8 all these days runs in 6 minutes on my machine-- which isn't particularly fast and since it's a Mac doesn't benefit from ccache on the gcs.  This compares favorably to the amount of time to start GCS, hookup a flight board, flash firmware, make any appropriate config changes/import, and stage for flying.  Unless you're actually iterating on a piece of functionality, I think clean builds are a false economy.

git also has multiple working copy support now, so there's much less cause to toggle back and forth between branches than there used to be.

PR #2112 (2015-12-24): uavobjgenerator:  Ensure rebuild of flight when necessary
This code had one remaining weakness-- on non-windows platforms, if you edited a file and then put it back UAVobjects might not get rebuilt due to moving to an "old" timestamp.  Now we touch the version file to ensure that this is OK.

From https://github.com/d-ronin/dRonin/pull/24.

Fixes #2110
closed 2015-12-24 by kubark42