TauLabs/TauLabs #2115: zero gyros while arming not functional on naze

ernieift, 2015-12-24:
When you move the vehicle while powering up, you get gyro offsets. This is normal, but the zeroing function should override these. On naze32, the gyros went zero when arming but grow to the old value and stay there in armed state.
Can't tell if it's a naze32/f1 only problem. On CC3D and all others is wasn't before. Have to research. It would help, if somebody test this behavior.

mlyle, 2015-12-24:
Ooh, this could explain some stuff I've seen on brushed quads.

kubark42, 2015-12-24:
Odd. The complementary filter should definitely zero out the biases during the arming procedure.

ernieift, 2015-12-24:
The reason is to be searched within smallf1/attitude.c, i guess. Zeroing is quite different here.

ernieift, 2015-12-24:
If i disable the function "zeroing while arming" and give the vehicle time to calm down. The gyros are ok.
Now the joke. I enable "zeroing while arming" and let the vehicle calm down after powering up. I look at the gyros and they are nearly zero. Now i arm it with delay. Yaw gets 3deg/s and the others also get offsets.

kubark42, 2015-12-24:
@ernieift Any chance you could do a rough bisect? I'd like to know if this worked, say, back around springtime.

mlyle, 2015-12-24:
There's not really any change to the F1 attitude thing since 2014.

This is the extent of the diff:

https://gist.github.com/mlyle/6537dd1744cabe160f09

A function is renamed, the file is renamed, and the TrimFlight stuff is removed.

ernieift, 2015-12-24:
At the end, the problem always sits in front of the screen. It took quite a while to find out what it was. The code is in right. The stored GyroTempCoeff [] in the Sensor Settings were not correctly and lead to miscalculations after zeroing the Gyros. The ZGyroTempCoeff on my Naze32 is -1.7648(deg/s)/deg and produced a noticeable deviation.
I let the Naze sit for a few minutes after power on and then again performed "Automatic Level Detection". It is important to press the save button, since the values obtained are not automatically saved.
Now the Tricopter stands still in the gcs.