TauLabs/TauLabs #2122: GCS not building since 06 Jan 2016

blaklite, 2016-01-17:
I pulled the latest "next" version via git, and tried to kick off "make package", which is how I usually do such things, and left my slow machine overnight to do its thing. But in the morning It seems that 99% of the work was done, ie flight code and most of GCS compiled, but it seems to have freaked out during producing some translations (namely for DE).

I had a quick look at Jenkins and noticed that there was no full build since 6 Jan.

Is this by design ?

kubark42, 2016-01-18:
Huh, no! Our automated builds are fully functional here, but we're a commit or two behind latest. Jenkins seems to be down again, @lilvinz could you kick it?

In the meantime, I'll investigate and see if I can find any differences between our internal branches and the external upstream.

blaklite, 2016-01-18:
Sorry, don't try too hard, but I just tried it with "make package V=1" and it seems not to have NOT crapped out at the last minute, getting past the DE translations, where before it was reporting "File Not Found ... taulabs_de.qm"

go figure

QT55~1.0/Tools/MINGW4~1/bin/mingw32-make -f Makefile
C:/TauLabs/tools/QT55~1.0/Tools/MINGW4~1/bin/mingw32-make -f Makefile.Release
/C/TauLabs/tools/Qt5.5.0/5.5/mingw492_32/bin/lrelease C:/TauLabs/ground/gcs/share/taulabs/translations/taulabs_de.ts -qm taulabs_de.qm
Updating 'taulabs_de.qm'...
    Generated 277 translation(s) (215 finished and 62 unfinished)
    Ignored 1183 untranslated source text(s)
/C/TauLabs/tools/Qt5.5.0/5.5/mingw492_32/bin/lrelease C:/TauLabs/ground/gcs/share/taulabs/translations/taulabs_es.ts -qm taulabs_es.qm
Updating 'taulabs_es.qm'...
    Generated 246 translation(s) (186 finished and 60 unfinished)

closed 2016-01-18 by blaklite

mlyle, 2016-01-18:
I don't think Jenkins is down; just no one has committed anything since Jan 6.

kubark42, 2016-01-18:
@mlyle The jenkins that does test builds is down. It needs a kick.

Out of curiosity, have you seen [Jervis](http://sam.gleske.net/jervis/)? This seems like a good way forward for allow more self-service for Jenkins.

mlyle, 2016-01-18:
Huh-- It seems like I can connect to it successfully and I see artifacts for the last merge and the last PR was tested.  Am I missing something?

Something like Jervis is great if you have a free-flowing environment where you need to create a lot of CI jobs.  At dRonin we just publish our jenkins config, which is pretty small and minimal compared to what was at Tau because of the packaging changes.  https://github.com/d-ronin/dRonin/wiki/Release-process

kubark42, 2016-01-18:
https://github.com/TauLabs/TauLabs/pull/2119 doesn't show it was tested. The PR test Jenkins server machine is a separate machine from the "nightlies" Jenkins server machine.

Publishing the jenkins config is cool, thanks, we'll take a look at that. Jervis I see as maximizing self-service for as much of the dev cycle as we can support. For instance I can't fix a broken script from here, I have to wait for @lilvinz to have some spare cycles to look into it.

mlyle, 2016-01-19:
Yah, we fixed that by basically taking the packaging scripts, etc, out and incorporating them into the build system.

It looks like it was down during #2119 but #2120 was tested (and failed).