TauLabs/TauLabs #2155: Consider switching from file-scoped copyright headers

peabody124, 2016-01-27:
Maintaining file-scoped copyright headers is a fairly outdated practice for large projects. Version control provides the accurate attribution of individual changes to individuals. 

See this document for more information https://softwarefreedom.org/resources/2012/ManagingCopyrightInformation.html

It might be worth considering moving to a consolidate project wide copyright (including attributing other projects that provide code). Historically (3-4y ago) we went so far as to add individuals to files but that wasn't useful and moved to the project system we have now.

peabody124, 2016-01-27:
ping @lilvinz since you did a lot of work standardizing the copyright template in files, what do you think about this?

kubark42, 2016-01-27:
That seems reasonable. I think I've read that header files can't even be copyrighted, so reducing useless, toothless boilerplate would certainly be valuable! I'm gonna love the LOC reduction on that one. :D

mlyle, 2016-01-27:
>  I think I've read that header files can't even be copyrighted

Files specifying only the calls involved in an API may or may not be copyrightable.  But headers often include a lot of other stuff, like documentation, outright code, macros, etc...

I have a couple of points of concern with this-- this would effectively lose all project-level attribution.  The "bits we've changed" that Tau has taken would no longer be apparent-- indeed who the individual copyright holder is for a block of code could be very difficult to untangle; was code that @tracernz committed from the time he was working on Tau or at dRonin?  The history is effectively stripped out.  Depending on many details of this, this may or may not be compliant.

I also think this will create a bunch of diff that will make it very difficult to move code between projects in the future; commits will not come cleanly that happen to touch the copyright block in dRonin or Tau (and a copyright block is still suggested to be present by that document), and it may trigger more complicated clerical actions in response to changes where top-level attribution information needs to be updated.

e.g. if you intend to import code from dRonin in the future, doing something like this may reduce your short-term pain but it is likely to be more problematic in general.

peabody124, 2016-01-27:
It maintains author level attribution with the granularity of what was done (i.e. commit history). It would _expand_ project level attribution to indicate that there is shared copyright with dronin. Also, file-scoped headers don't indicate what project someone was working with when they wrote a particular commit when there is simply a line that applies to a certain file. You would need something in the commit message itself to indicate that (or ick - comments in file like we used to have, ugh).

As for commit conflicts, those are really easy to fix when cherry-picking.

kubark42, 2016-01-27:
Agreed with @peabody124. In theory @tracernz can submit code to OpenPilot and the authorship line wouldn't make that clear, whereas the SCM commit is much more authoritative.
 
> I also think this will create a bunch of diff that will make it very difficult to move code between projects in the future.

That's possible. I've seen git choke on less. It's an argument for achieving alignment before heading too far down this path. On the other hand, authorship attribution lines has historically only been used as a stick to poke in someone else's eye, so I'm all for resolving this problem before it bites again.

Let's keep in touch on it.

> As for commit conflicts, those are really easy to fix when cherry-picking.

I've had issues with this in the past, plus it's more resource intensive. If we could find a better way that would be nice.

mlyle, 2016-01-27:
> On the other hand, authorship attribution lines has historically only been used as a stick to poke in someone else's eye, so I'm all for resolving this problem before it bites again.

I want to be clear that's not where we're coming from.  If you look at our PRs, issues, (issues opened before leaving Tau), etc... this is very, very important to us.  Good hygiene here is critical; we want vendors to be able to do stuff with the code (heck, I might want to in the future, myself) commercially without undue exposure.

I'd really favor an approach where we have the file level ones and the person who takes a PR from the other project adds a commit adding copyright notices as necessary.  As time goes on, this will get less burdensome as the notices end up in more and more files-- I hope we'll both be tolerant if a case happens where a date is inadvertently not bumped.

mlyle, 2016-01-27:
(It might be in everyone's interest if we were to move to that kind of top-level attribution, and the file-level thing isn't my favorite, in general.   But-- dRonin is not equipped to make that move ourselves.  We are simply not familiar with enough of the history to safely make that type of wholesale change in notices without possibly coming out of compliance ourselves.  And enough has changed / drifted that we wouldn't safely be able to apply yours.  I think it's important or beneficial for both projects to keep using the same approach as long as there's going to be significant flow of code between them).

peabody124, 2016-01-27:
Dronin certainly doesn't have to make the same change if that isn't what you guys want. Historically, almost all CR bumps have ended up in separate commits (normally from either a PR review, or for example at DR as a bulk change), so I really wouldn't anticipate this being much of a barrier to code exchange. It will make our CR more consistent, accurate, and avoids any potential of discrediting DR which I like.

Note: bumping dates is equally important for doing the right thing. Similarly to even when you guys haven't added the new CR to the header and expect us to add it, you should be bumping the TL date when going from our repository.

mlyle, 2016-01-27:
> Note: bumping dates is equally important for doing the right thing. Similarly to even when you guys haven't added the new CR to the header and expect us to add it, you should be bumping the TL date when going from our repository.

We'll expend effort to bump dates (and have).  We're glad to look the other way when you miss a notice accidentally.  But when it's systemic like this it's a problem.

> Similarly to even when you guys haven't added the new CR to the header and expect us to add it

AFAIK all of the files you've cherry-pick'd the changes from have our copyright notice in the current repo.

mlyle, 2016-01-27:
Here is what I would hope to see if Tau were to go this direction:

* A top-level project copyright document that prominently lists dRonin (at the top of the block, with Tau and OpenPilot) as a copyright holder in the combined work.
* File specific headers that note the copyright information is available at __url__
* Future commits should include the dRonin PR# in the commit message so that the project-specific metadata and history are not lost, as opposed to just retaining the individual commits' github attribution information (as that commit does not exist in a vacuum).    e.g. 'from d-ronin/dRonin#123' at the end of the subject.

Even though I prefer this centralized top-level approach, I discourage it, as it's going to cause significant problems in cross-project flow both directions-- both changes you guys need to snip off us, and a significant mystery for us on how to attribute anything we import from Tau.  We've held off on big code reformats and reorganizations to maintain some compatibility both ways, but if this is going to be stripped we're likely to just reformat and diverge.

kubark42, 2016-01-28:
Another reason to get rid of the boilerplate is it needlessly confuses project-wide text searches. For instance, searches for the number ``11`` cause every file in the project to show up, because the zip code for the Free Software Foundation is ``02111``. Likewise for many keywords which show up in the GPL preamble. I understand replacing it with a short form won't solve the problem, but it reduces the probability of collisions so it's a net positive to get rid of the boilerplate.

Having given this a day of thought, I'm fully in favor of transitioning as long as it's legally compatible with the licensing requirements. It will avoid futile conversations about authorship lines, decrease LOC, and increase the utility of IDE tools. 

I'm going to [drop by the FSF](http://www.fsf.org/about/contact/visiting-the-free-software-foundation) and ask them what they think.

mlyle, 2016-01-29:
Just note that this approach is now incompatible with further changes to dRonin since d-ronin/dRonin#558.  So you guys may be able to cheat us out of credit this once through a top-level licensing approach, but it's not going to work in the future.

peabody124, 2016-01-29:
You can't actually relicense a project without permission of all the authors. Either way we are making this change to make sure to credit DR. Not sure why Dronin is so intent on creating drama when we are actively going above and beyond to give credit and preserve PR references in pulls.

mlyle, 2016-01-29:
We're not relicensing a project.  We are licensing all further changes with that additional term.

peabody124, 2016-01-29:
Pretty sure you cant relicense anything derivative on GPL without all authors approving, so it wouldnt affect anything based on existing code

mlyle, 2016-01-29:
It's explicitly permitted by the license to add a further restriction as listed above.

Things get licensed "more tightly" all the time.  A lot of GPL code is past-BSD licensed code.  The intention of those clauses is to allow commercial distributions of existing GPL projects with reasonable commercial restrictions, e.g. indemnification requirements.

This is really pretty simple:

```
Notwithstanding any other provision of this License, **for material you add to a covered work**, you may (if authorized by the copyright holders of **that material) supplement the terms of this License with terms:

a) Disclaiming warranty or limiting liability differently from the terms of sections 15 and 16 of this License; or
**b) Requiring preservation of specified reasonable legal notices or author attributions in that material** or in the Appropriate Legal Notices displayed by works containing it; or
```

kubark42, 2016-01-29:
I plan to head over to the FSF and get their opinion and report back here. They're the most knowledgable people I can think of. I'll also get clarification on whether replacing the GPL header with a header that specifically states where to find the GPL header is a sufficient legal notice. And if we don't listen to the FSF on this, then we really don't need to be arguing the point.

It seems to me there's an attempt to claim a moral issue as well, but it's hard to get to the meat of the matter because it's been obscured by the bullying accusations of "cheating", "stealing", DMCA threats, etc...

peabody124, 2016-01-30:
Thanks @kubark42. I'm sure the issue of how to properly port commits between forks is pretty well resolved as well as how a copyleft license can be further restricted without permission of all holders.

mlyle, 2016-01-30:
> It seems to me there's an attempt to claim a moral issue as well, but it's hard to get to the meat of the matter because it's been obscured by the bullying accusations of "cheating", "stealing", DMCA threats, etc...

It's really simple.  You can take out the code you took from the dRonin project, or you can maintain the copyright notices that were attached to it.  You've said in the past maintaining these was something you intended to do.

The contributors to dRonin are agreed on this point-- that using our work involves preserving our copyright notices on that work.

Is this something you intend to do or not?  Right now you are out of compliance with the license agreement.

> Thanks @kubark42. I'm sure the issue of how to properly port commits between forks is pretty well resolved as well as how a copyleft license can be further restricted without permission of all holders.

It is, after all it's right there in the license text as one of the permissible further restrictions :)

tracernz, 2016-01-30:
I fail to understand why you are so opposed to preserving the copyright notices? AFAICT it's been the standard practice at this project since the beginning, as evidenced by the headers on almost all of your files. Failing that it's just a common courtesy. While it's understandable they could be easily missed, it's ridiculous to not comply when it's been pointed out.

mlyle, 2016-01-30:
Yah.  We don't expect perfection, just a good-faith effort to comply with the license terms, and a prompt correction when you're out of compliance.

peabody124, 2016-01-30:
@tracernz this entire issue is about making sure we don't fail to attribute DR. Also see my comments above about how those headers have evolved over time as is. There wasn't ever a point in time where we actively sat down and decided per-file is the best way to go (at least in the 5 years I've been involved) and having read about it per @mlyle's prompting was what made me realize this isn't considered best practice for large projects. So this change both avoids failing to attribute _and_ makes maintenance easier. Win Win.

mlyle, 2016-01-30:
@peabody124 Point of clarification-- would you be OK with us removing the Tau Labs copyright notices throughout the dRonin source?  We can add it as a footnote in the toplevel license file that some of the work came from you guys and is copyrighted by Tau Labs.  It'll be a maintenance win for us, too.

In the meantime, you're presently out of compliance and should fix that.

mluessi, 2016-01-30:
@peabody124 @kubark42 can you explain why you want to change the current practice? Obviously, one of the copyright holders is not happy with the change and I think it would be best to abandon this plan.

Personally, I like having file based copyright headers, as it lets you see where a particular piece originated from.  With the single copyright file approach, you could basically have one file saying "parts of the code came from: Linux, dRonin, ... (list any project you can think of)" and it is then impossible to know where code came from without combing through the git history (and the git history is not part of the actual copyrighted material).

peabody124, 2016-01-30:
@mluessi to reiterate the points above:
1. see that link indicating project wide notices is a preferred practice for larger projects
2. reduced boiler plate code
3. make sure we don't fail to attribute to DR
4. less maintenance
5. to actually seem meaningful contributions only git history (e.g. git blame) is useful. having a 1 line change that with also a copyright inclusion is hardly the same as a file being 99% written under TL, so the copyright itself provides no utility for attribution

@mlyle sure, a project-wide copyright at dronin would be fine indicating that it is a shared copyright amongst 3 projects. making it a footnote seems fairly inappropriate / legally questionable and isn't what is suggested here. For the record Dronin is hardly on a high moral group for attribution give your home page gives no indication the code is fork from TauLabs and talking about "Dronin's advanced Autotuning" as if it wasn't written by me with TL ;-) You just don't see us spending our lives making drama.

@kubark42 while talking to the FSF ask for clarification on the implications of https://github.com/d-ronin/dRonin/pull/558 .  I'm pretty sure there are none since they can't change the license on derivative code without agreement of all the copyright holders, but it would be good to clarify.

mlyle, 2016-01-30:
> clarification on the implications of d-ronin/dRonin#558

Go ahead-- Ken, ask your corporate counsel whether you can ignore #558.
:).  Be sure to remember the nuclear implications (permanent license
termination) associated with a violation of such a restriction.

If you deliberately violate the license agreement (I understand small
missteps will happen) I will pursue all available remedies.

End of conversation.

peabody124, 2016-01-30:
They would probably point out this line from the GPL
```Notwithstanding any other provision of this License, for material you add to a covered work, you may (if authorized by the copyright holders of that material)  ``` and point out that adding that was authorized by all the copyright holders. And that by making the change suggested in this PR we are intentionally making sure we have "preservation of specified reasonable legal notices or author attributions " by moving it to a consolidated location. See the discussion above from the link above that states moving the copyright information around is permitted.

Not sure why you are so passionate about being dramatic when we are trying to make sure to attribute you.

mlyle, 2016-01-30:
"That material" refers to "material you add to a covered work". Indeed
cyphy is likely to make use of that exact term to tack on limitations of
warranty, etc, beyond what the gpl states.

James, I wish we could assume good faith.  But our upfront request for
preserving these notices as required was ignored.  In response to us asking
for this to be cured, you've proposed breaking with years' long practice in
how the notices are preserved.  This has upset many people that their work
is no longer credited, and complicates dRonin's legal stance if we ever
need to prevent commercial abuse of our project's work--- since the code is
now being distributed without our copyright notice.

Maybe I'm being dramatic, but several people are upset about this-- many of
whom are not people I would characterize as dramatic at all.

DMCA notifications have been sent to github for both tau's and cyphy's
repository for a few specific PRs.  When they are processed-- presumably
monday-- you will need to decide whether to cure the violation, send a
counter-notification stating under penalty of perjury that you do not
infringe, or to allow github to take down the repository within 24 hours.

kubark42, 2016-02-01:
DMCA threats and abusing the system's knee-jerk reaction to DMCA takedown requests is just churlish. I'm sorry the rest of you are getting yourself dragged along with this bullying. Both the Tau Labs and LibrePilot founders saw how a loud voice caused a community to be mean and spiteful, please be cautious you're not taking part in a variant of this. 

I spoke with the FSF this morning. It was a nice conversation. The TLDR; is that they understand the value, but "It's not clear you should move the warranty statement off of the file. There are cases to be made for being able to do it and people like the SFLC can help suggest how you would go about doing it". So I don't really think there's anything important enough in this to warrant bugging people in a more formal-- i.e. legal-- setting.

The FSF's opinion is that it's best to be as clear as possible so that there is the minimum of legal expert witness required if a company is ever brought to court for GPL violations. They admit the value of SCM in the authorship conversation, but don't feel that changes the basic equation. Totally fair point. So as far as I'm concerned, placing files into legal ambiguity isn't worth the benefits. 

Couple interesting things of note:

1. The SFLC is the FSF's legal council. That means that the opinion authored in the original link is about as authoritative as they get.
1. I don't think anyone has signed a CLA (Contributor License Agreement). That means that the idea of project code ownership-- as opposed to individual code ownership-- is murky. I'm paraphrasing here, but here's the note I took: "If making non-trivial modifications, the authors should have updated the authorship lines individually, unless there’s a CLA."
1. The FSF does not recommend removing the boilerplate from new files. They want the strictest view on this because they want everything to be done correctly so nothing is left ambiguously after the fact. (Indeed, the gal said jokingly that "From a legal perspective, you should print the whole GPL and physically mail it to every person who downloads the file!")
1. We should upgrade to GPLv3+. The primary motivation for this is to allow the work to be upgraded to GPL v4, should that version ever exist. (No current plans).
1. The addresses in the file template are wrong and should be updated. The FSF has moved and their new address is 51 Franklin St, Boston, MA 02110.

mlyle, 2016-02-01:
@kubark42 -- What did they say about our ability to add that paragraph requiring maintenance of the file-level headers-- did you get around to that question?

Indeed, the Copyright Tau Labs in all of the files is somewhat meaningless because there is no defined "Tau Labs" legal entity.  Ideally OmniLoco would secure a DBA or some other means to make the name an actual legal name, though it's unclear the status of all the things that have had copyright notices added before that time.

It's true about CLAs and that ideally copyrights should be transferred with an appropriate writing.  The FSF is exceptionally careful with this (and with securing employer releases and other measures), but there's a wide continuum in practices across different open-source licenses.

> DMCA threats and abusing the system's knee-jerk reaction to DMCA takedown requests is churlish. I'm sorry the rest of you are getting yourself dragged along with this bullying.

Man, I guess you missed me a few days ago saying that I intended to hold off a bit longer on IRC and the chants of "Just DMCA them already!" :p 

I don't know how the process goes from here; when you get the DMCA notice, assuming that PR is landed I'm glad to tell GitHub the issue is resolved.

In other news, there's a number of Makefiles that are pretty clearly copies of OpenPilot makefiles (e.g. sparky2; they should have the openpilot copyright line added back in.  We'll be doing that over at dRonin, but Tau probably should to be in compliance as well.  There is also #1998 which has put both projects out of compliance.  Can we work together on these issues and identifying anything else we need to have good licensing hygiene?  Since you're a commercial user and redistributing this code, it seems especially critical from your standpoint that all of this is handled well.

The GPLv3 is mostly a good license, but it has a rather nuclear consequence on license violations-- termination until reinstated by the author (e.g. no right to redistribute afterwards, even after cured, depending on circumstance).  We need to be very careful.

mluessi, 2016-02-01:
In my opinion, the DMCA takedown request was a bit of a drastic measure to resolve this situation. On the other hand, I feel that we would probably not have reached an agreement without it as the legal ramifications of removing the headers are somewhat unclear and therefore the discussion would probably have dragged on forever.

Once #2172 has been merged, the DMCA takedown should be baseless and not cause any problems for TL. Hopefully we can all move on and continue to collaborate.