TauLabs/TauLabs #2215: Objects not saved FlightBatterySettings

HenryLeinen, 2016-03-13:
I am using a Quanton board with the latest software (Nov. 2015).
I activate the "Battery Monitoring" Module --> it works in general.

However, whenever I try to change any of the parameters (e.g. the number of cells or the battery capacity) and click on SAVE, I get a negative feedback and in the log it says :
20:37:03[DEBUG][smartsavebutton.cpp] Sending object to remote end -  "ModuleSettings"
20:37:03[DEBUG][smartsavebutton.cpp] Save request for object "ModuleSettings"
20:37:03[DEBUG][smartsavebutton.cpp] Sending object to remote end -  "FlightBatterySettings"
20:37:03[DEBUG][smartsavebutton.cpp] Save request for object "FlightBatterySettings"
20:37:03[DEBUG][smartsavebutton.cpp] failed to save: "FlightBatterySettings"
20:37:03[DEBUG][smartsavebutton.cpp] Error, tried to apply/save a non-settings object
20:37:03[DEBUG][smartsavebutton.cpp] Sending object to remote end -  "PicoCSettings"
20:37:03[DEBUG][smartsavebutton.cpp] Save request for object "PicoCSettings"
20:37:04[DEBUG]RESULT "Objects not saved:\nFlightBatterySettings\n" () ("FlightBatterySettings") ()

~LeiHen();

peabody124, 2016-03-13:
You try power cycling after enabling the module

HenryLeinen, 2016-03-13:
That does not work. Tried it several times. It always gives the same error. After power cycling, the same 'initial' values reappear.

kubark42, 2016-03-14:
@HenryLeinen Is it possible you have a firmware/GCS mismatch? That would explain this error.

HenryLeinen, 2016-03-23:
I checked it, but firmware and GCS are matching (both are on 20151123).

HenryLeinen, 2016-04-08:
I have deleted the settings partition, reconfigured everything and now it works. I have double checked on my other quanton board. Everything works. I think we can close this issue.

closed 2016-04-08 by HenryLeinen