TauLabs/TauLabs #2233: OpenPilot's proud policy of censorship continues here @ TauLabs, through rewriting history at https://github.com/TauLabs/TauLabs/pull/2231 and deleting all comments by contributors to other proejcts

mlyle, 2016-03-18:


kubark42, 2016-03-18:
And this is constructive because...?

Ultimately you were doing net negative work by refusing to honor our requests to stay on topic. We did what every community does when faced with this level of noise: we asserted our authority over our own project to keep the message technically on topic. When faced with the unfortunate choice of having to ban a member or deleting their messages, we chose to delete the off-topic messages as a banned member has no voice anymore whatsoever.

We demand a higher level of maturity in our dialog.

closed 2016-03-18 by kubark42

mlyle, 2016-03-18:
OK, so please preserve the history instead of reciting your side of the story.  Link / attach the screenshot, along with whatever lasting comments... provide a venue for discussion, etc.

I think it's really strange that you asked other people to speak up, and then when several people did, you deleted the entire discussion and posted a jab at me.