TauLabs/TauLabs #2235: wdg_configuration.bootup_flags = RTC_ReadBackupRegister(PIOS_WDG_REGISTER);

tofly1990, 2016-03-18:
HI 
    When System ReBoot caused by "SYSTEMALARMS_REBOOTCAUSE_INDEPENDENTWATCHDOG",the flag"bootup_flags"alaways be "0"; I review Code ,and think it not right; I can not konw which Task let watch dog overflow.
     Can anyone helpe me? Thankyou!

tracernz, 2016-03-18:
Related to this? https://github.com/TauLabs/TauLabs/pull/2228 Make sure you're running the latest if you're on next.

tofly1990, 2016-03-19:
I don't know, My board(Sparky2) always ReBoot An hour later(about 69 min);Cloud anyone Test It?

tofly1990, 2016-03-19:
I Test many times,The results are the same.

tracernz, 2016-03-19:
Oh yeah, known issue. It's 4295 seconds/1hr 11 min on all boards running ChibiOS.

tofly1990, 2016-03-19:
HAHA, So it is! Thankyou, I think developer can quickly resolve the problem.

mlyle, 2016-03-19:
@tracernz nice to know this isn't particular to us.. I mentioned this to @peabody124 shortly after we forked and I think he thought it was specific to us.