TauLabs/TauLabs #2236: Taulink module R3 value is incorrect

AlessioMorale, 2016-03-22:
(Not sure this is the right place to raise this issue, though)
R3 for taulink module shall be raised to guarantee enough headroom for the LDO to work properly. With BOM values the switching output is only 3.3V while it should be 2-400mV higher than that.

peabody124, 2016-03-22:
Thanks for reporting this! I'll look into it / check what I populated / measure some voltages and post back / fix the schematic. Are you using the hardware? Liking it?

AlessioMorale, 2016-03-22:
I've finished to assembly it yesterday and seems to work fine. Sooner or later i'll modify it to add the other inverter on s.port to also support crossfire protocol with Taranis.
Overall it looks nice!

AlessioMorale, 2016-03-22:
Another option may be to use a 48 pin low end stm32F3. The existing inverter logic may then be removed altogether.