TauLabs/TauLabs #2238: PPM not working on Brushed Sparky

Swirklich, 2016-04-01:
Tried two different d4r-ii RXs in PPM mode (signal pin 3 + 4 bridged) that I swapped back and forth between another quad (works fine) and the Brushed Sparky (doesn't work at al). d4r-ii is bound to my Taranis (solide green light), but like I said: the same RXs work on my other (non-TauLabs) boards.

Used a multimeter to check continuity between Signal from the RX  through the RCVR port and the CPU (all pass). 
Tried the former and latest configuration Brushed Sparkt configuration files from GitHub. In GCS, the PPM signal isn't pick up at all. Hardware -> RCVR is set to PPM. Power-cycled multiple times, settings appear to be saved (didn't default to Disabled). I also verified this under ReceiverActivity. Neither the wizard nor manual configuration pick up anything.

Swirklich, 2016-04-02:
As mentioned by another user: 

_I had similar issues but with a MicroFrX. It seems like mine were caused by the combination of a series resistor in my receiver and the pull-down resistor on the Brushed Sparky interacting to cause the PPM signal to be attenuated too far. If you have a scope, try to probe the PPM output from your receiver when it's not plugged in to the brushed sparky, and when it is. If you see the PPM signal clearly when it is not plugged in, but it gets squished down when it's plugged in to the Sparky, you have the same issue as I did. My fix was to remove the 10k pull-down resistor on the Brushed Sparky._

I hope we can work around that resistor, as it would be a real pain for every PPM user to remove a tiny resistor on the PCB (this could go wrong in so many awful ways with untrained hands).

peabody124, 2016-04-02:
I wasn't able to replicate your issue. Using next PPM input works on my BrushedSparky:

[image]

Here is the configuration:
[image]

This is my connection:
[image]

Swirklich, 2016-04-02:
Thanks Peabody, is there anything else I can try? I have been wrecking my head around this for a week now, trying different d4r-ii's... What am I missing?

Swirklich, 2016-04-03:
Thanks for taking the time, Peabody. I am out of ideas. You said you've had your d4r-ii for a while now. I know they changed their firmware a couple of times over the past year or so on the RX and this one is only a few weeks old. I can't be positive this _is_ the reason, but we've pretty much excluded everything else by now. 
**Could you do me one last favor please and point me towards the 10k pull-down resistor on the below schematic** (see my 2nd post above):

[image]

Another user had these same issues not too long ago, also with a FrSky RX, and I am willing to try the same if know which one it is exactly.

Swirklich, 2016-04-03:
Flashed Raceflight, everything functioning but PPM input. Peabody, I think you can save yourself the trouble of trying latest current release. This _must_ be hardware.

**Please help me locating 10k pull-down resistor on Signal so I can remove it and see if this resolves issue!**

peabody124, 2016-04-04:
[image]

Here it is. If you have verified the continuity between the receiver pin and the MCU that would be the only hardware error I could imagine that would cause this. I'm skeptical the resistor is the problem, but possibly it is. Sounds like I'm using the same receiver you are.

Swirklich, 2016-04-05:
I think I just got unlucky and received a dud. Removing the resistor didn't change anything. Oh well, sorry for wasting your time mate!

closed 2016-04-05 by Swirklich

JR63, 2016-04-07:
from peabody124: "that would be the only hardware error I could imagine that would cause this"

There could be another hardware error: If the MOSFET (Q7A) is defective (gate - source shorted) it could result in pulling the PPM signal down.

Try to desolder the Q7 and give it a try ;-)

Swirklich, 2016-04-07:
JR63, thanks a lot man. That did it indeed! You raaawk :). Let's if everything else behaves normal now!

JR63, 2016-04-07:
:-)

peabody124, 2016-04-07:
Great catch!!! Let me check the code and verify the gate input is actively driven in all cases.

Swirklich, 2016-04-07:
Unfortunately, PPM was erratic and all over the place, Sparky was uncontrollable. Tried a second brand-new d4r-I. 
Today, while doing some tests, my Brushed Sparky finally gave up the ghost and let go of the magic smoke when I plugged in the battery once more. No idea what happened. Which means that, for better or worse, I am out of the game for good now.

JR63, 2016-04-07:
Bad news :-(