TauLabs/TauLabs #2243: REVO

Tomcatt, 2016-06-09:
Hello,

I see you support the REVO mini, but am I able to use this on the full REVO?

buzzdev, 2016-06-14:
Hi @Tomcatt i use it on Revo Mini and also on Full Revolution.
It works on both.

mlyle, 2016-06-14:
@tomcatt / @buzzdev Note that Tau Labs doesn't seem to be getting updated anymore, and we've done significant things to improve Revolution support in dRonin ( http://dronin.org )

Tomcatt, 2016-06-15:
Woo, bro something new? I'm not home right now to download it. Let me see if I can download it.

Okay if TauLabs is dead, LiberPilot still gets updates here and there and not really wild about PID tuning on RaceFlight so I came back. 

I like testing new things. I'll give it a try.

buzzdev, 2016-06-15:
@mlyle Yes, i know. In fact i am using dRonin on both FCs, while still tempting RaceFlight.
His question was, if TauLabs runs on REVOs, which it does.

It seems, that it is still "being worked" on TauLabs though:
`peabody124 committed on Apr 30`

Tomcatt, 2016-06-15:
Do you guys have really good community here?

mlyle, 2016-06-15:
I think so.  http://forum.dronin.org

buzzdev, 2016-06-15:
...signed up... 👍

Tomcatt, 2016-06-16:
Signed up as well