TauLabs/TauLabs #2247: "recipe for target 'uavobjgenerator' failed" Error 127

mettinoktay, 2016-07-21:
`make all` gives this error:
`*NOTE*      Sanitized disallowed variable 'TMPDIR' from environment`
`/usr/bin/sh: line 1: qmake: command not found`
`makefile:317: recipe for target 'uavobjgenerator' failed`
`mingw32-make: *** [uavobjgenerator] Error 127`

peabody124, 2016-07-21:
Run '''make qt_sdk_install'''

mettinoktay, 2016-07-22:
I guess something very essential is missing. "make qt_sdk_install" gave the same error.

$ make qt_sdk_install
*NOTE*      Sanitized disallowed variable 'TMPDIR' from environment
/usr/bin/sh: wget: command not found
C:/Users/mettinoktay/taulabs/make/tools.mk:87: recipe for target 'qt_sdk_install' failed
mingw32-make: *** [qt_sdk_install] Error 127

mlyle, 2016-07-22:
Follow this development setup procedure:  https://dronin.readme.io/docs/development-setup-windows

This documentation is for dRonin, but it's very similar to the procedure for Tau Labs.

peabody124, 2016-07-22:
For Tau Labs you might want to look at this https://github.com/TauLabs/TauLabs/wiki/Development-Windows-setup

tracernz, 2016-07-22:
The Tau qt_sdk_install target doesn't work because it depends on wget and that isn't included in git for windows anymore. Change the wget line in make/tools.mk for curl and it will work.

mettinoktay, 2016-07-25:
Hey guys I switched to Ubuntu and everything compiles now.

closed 2016-07-25 by mettinoktay