TauLabs/TauLabs #2249: What flight controller can I use

brendabrandy, 2016-07-22:
I am currently working on brushed sparky2, and I was wondering what kind of flight controller I can use. I know that I can connect taulinks to android GCS, but is there another physical flight controller that I can configure?

Thanks!

peabody124, 2016-07-22:
Are you asking what flight controllers are supported by Tau Labs? You can see it on the wiki https://github.com/TauLabs/TauLabs/wiki

* [[Flying F3|Creating-a-FlyingF3-from-scratch]] - cheap (<15$) commercially available STM32F3 based platform system.  Typically requires a shield to break out the connectors or some manual wiring. Capable of RTH and PH.
* [[Sparky]] - small single sided STM32F3 flight controller. Capable of altitude hold, RTH and PH. [more information here](http://buildandcrash.blogspot.com/2013/05/sparky-testing-and-building-no-crashing.html)
* [[Sparky2]] - more powerful STM32F4 version of Sparky with integrated radio. Capable of altitude hold, RTH and PH and navigation. [more information here](http://buildandcrash.blogspot.com/2014/08/sparky-20-and-taulink.html)
* [[RevoMini|RevoMini]] - OpenPilot STM32F4 based board.
* [[FlyingF4|Creating a FlyingF4 from scratch]] - STM32F4 based discovery board but needs a shield to provide sensors.
* [Quanton](http://www.quantec-networks.de/shop/en/quanton/1/quanton-flight-control-rev.-1) - powerful STM32F4 based platform with large amounts of connectivity including 8 PWM in and 8 PWM out, and four serial ports.  Fully navigation capable.
* [Gemini](http://www.team-blacksheep.com/products/prod:gemini) - mini FPV hex racer from Team Black Sheep that uses Colibri flight controller (derivative of Quanton).
* [Copter Control](http://www.openpilot.org/products/openpilot-coptercontrol-platform/) - Older STM32F1 based platform.  Legacy support and not recommended for new designs.

brendabrandy, 2016-07-29:
@peabody124 :  I think I use the wrong term, I was thinking more about the transmitter. I currently have a taranis that supports open tx, can I use that on brushed sparky?