TauLabs/TauLabs #2253: DShot support

peabody124, 2016-11-12:
There is a new digital protocol floating around for ESC control called DShot. I'm not particularly convinced in of itself it will improve anything much functionally beyond OneShot125. However, what is particularly appealing about it is the ability to query the ESC the RPM:

http://www.rcgroups.com/forums/showatt.php?attachmentid=8524039&d=1450424877

This has been supported for F3/F4 on BetaFlight: https://github.com/betaflight/betaflight/pull/1282

What is _really_ appealing to me about it is the ability to query the ESC RPM. Feeding this directly into the LQG controller (#2237) would be a huge win as we wouldn't need to estimate the instantaneous torque. Running system identification to know the time constant would still be used to compute the optimal control parameters and the estimation would benefit even further using the true instantaneous torque to smooth out the gyro estimate better than the current kalman filter.

The down side is that the current telemetry feedback for RPM is still a 1ms packet. Given we only want the 2 bytes of RPM for that it might be good to strip it down if used in the control loop and not to display things like temperature.

mlyle, 2016-11-12:
I believe the way this is implemented you are supposed to chain a 4th wire between escs and a flight controller UART.  Every 4ms you make a telemetry request to a given ESC using dshot and get feedback.  Or you spend a lot of uarts or do some really clever bit of softserial and use other gpios for the return.   I've skipped it so far (doesn't even require dshot to make telemetry reqs to kiss ESC) because 125hz feedback doesn't seem that useful.

peabody124, 2016-11-12:
The documentation says it supports half duplex and I've already support for that ages ago with TL-ESC. 125hz feedback would still be useful information the LQG kalman filter can use and would allow detecting motor failures. Certainly not needed tomorrow to get in the air but will be useful for next gen control

peabody124, 2016-11-12:
Here is a link to the telemetry support for betashot https://github.com/betaflight/betaflight/issues/673