TauLabs/TauLabs #2259: Sparky2: PWM out/in simultaneous

peabody124, 2017-01-08:
Add support to Sparky2 for the 6th PWM output to become a PWM/PPM input. This will allow receiving the RSSI signal from something like an SBus receiver.

Side note, I'm surprised the RSSI is not available in the SBus data stream.

peabody124, 2017-01-08:
Workaround: https://oscarliang.com/rssi-ppm-channel-taranis/ shows how to use the Taranis to echo the RSSI signal from telemetry back out to a particular channel

tracernz, 2017-01-08:
> Side note, I'm surprised the RSSI is not available in the SBus data stream.

Seems to be common to most brands, quite weird.

peabody124, 2017-01-08:
@tracernz do you mean you have you seen anything suggesting it is available in an SBus stream, and only some brands don't have it?

tracernz, 2017-01-08:
No, by brand I meant first-party vendors like Futaba, Spektrum, Jeti who define the various protocols. :wink:

peabody124, 2017-01-14:
Ok. I'm going to actually close this as won't fix. Turns out the RSSI signal isn't a sensible PWM signal. You can either analog LPF it, or have the Taranis relay it as a looped-back channel and then import it from the SBus/PPM stream. I opted for that (although the failsafe behavior is a bit jerky and drives it to a nuts value, so it's still clear you lost signal)

closed 2017-01-14 by peabody124