TauLabs/TauLabs #2263: Can't arm my quad

BaronRanger, 2017-03-14:
Hello and greetings, 
I have a CC3D flight controller, all is well and I am good to go, The RC reading are correct and all that, but the problem is that I can't arm my quad, no matter what type of arming I set it too, I've tried Switch (which is the one I want it to be set on), Yaw right, Roll left and yaw right, ....
What's the problem then?
[image]

[image]

BaronRanger, 2017-03-14:
I would like to add that it won't even arm when it's set on "always armed"

tracernz, 2017-03-14:
That looks like http://dronin.org? Any system alarms? See https://dronin.readme.io/docs/getting-started#verify-settings-and-configuration

BaronRanger, 2017-03-15:
Hey.
I don't have any servos so i thought the Servo Output being red was normal, but then I configured channels for my motors and It all got fixed. 
Sorry for the pointless topic.
Thanks for your reply!